45clouds/WirelessCarPlay #1: More insights

janbrodhaecker, 2020-01-05:
Hi! Can you give some more insights on the idea of how this should work with a raspberry?

45clouds, 2020-01-07:
Hi Jan! I would like to put RPi3 into armrest base box. I have 2 USB ports there (both of them are CarPlay enabled). One of them will be directly connected to RPi3 using USB-A cable (male on both sides). Another one will be used as a power supply to RPI3. There are some presentations available on apple.com website that describe pairing process. Part1: https://developer.apple.com/videos/play/wwdc2016/722/ and part2: https://developer.apple.com/videos/play/wwdc2016/723/. The first step is to manage bluetooth connection between iPhone and RPi3 and after successful pairing the wireless connection should be established. Unfortunately bluezd stack doesn't support iAP2 profile. That's why I'm trying to implement part of Accessory SDK available from Apple to certified MFi hardware developers.

45clouds, 2020-01-07:
You can read this article: https://blog.csdn.net/huangke2012hk/article/details/51240615 (please use google translate).

45clouds, 2020-01-07:
And here is bluetooth part: https://blog.csdn.net/huangke2012hk/article/details/51226220

janbrodhaecker, 2020-01-07:
Wow! Thank you really much for the input! I am also really interested in achieving the same as you. I did not even had an idea where to start, but this helped me a lot. I will dive into the resources, you provided.

45clouds, 2020-01-07:
Ok. I've a small gift for you. Please look at carplay.pdf in this repo :)

janbrodhaecker, 2020-01-07:
Oh my god! Thank you so much! Just seems like the holy grail for me! Thank you so much!!!

janbrodhaecker, 2020-01-10:
> Unfortunately bluezd stack doesn't support iAP2 profile. That's why I'm trying to implement part of Accessory SDK available from Apple to certified MFi hardware developers.

Do you have already any idea where to start here?

45clouds, 2020-01-15:
Hi Jan,

I have to talk to this man. Please read: https://hackernoon.com/building-an-ios-hardware-app-401eb4869012

janbrodhaecker, 2020-01-16:
Hi! Thanks for your answer, I really appreciate that you are sharing your information on that topic! 

I just investigated a bit for an hour or so and I think without the accessory specification we will not get to manage it. To use iAP2 we would need to use the software authentication as we do not have any access to the Apple Authentication Coprocessor - but there is no documentation about that available for non-MFi  people.

45clouds, 2020-01-16:
Yes.... this topic is quite difficult as I see. I committed some new documentation directly from Apple. Look at Accessory Authentication (most important part), Authentication Coprocessor, Bluetooth Accessories and Device Authentication. Let me know if you will have some ideas because I'm a little bit lost right now.

janbrodhaecker, 2020-01-16:
Thanks! I see, this documentation is "free" available on the internet - I guess this is too old, as the the software authentication was introduced one year ago (as far as I understood it from the posts). 

We would need a more recent documentation, I guess.

janbrodhaecker, 2020-01-16:
Regarding the new version, maybe this link is helpful: https://download.csdn.net/download/SJno111/11984618

But I am not able to download ... and I am also not sure if this helps or not.

45clouds, 2020-02-05:
Hi Jan. I got this file. It seems that Apple removed whole Chapter 23 (CarPlay) from this document. Please contact me directly and I will share this file with you.

Vulpecula-nl, 2020-02-06:
I also interested in Apple Carplay on a raspberry pi. Do you have anything working to show?

45clouds, 2020-02-07:
Hi,

Unfortunately we are still in early stage. The most important and interesting part is to establish BLE connection between iPhone and Raspberry which is iAP2 authenticated. Normally you need Apple Authentication Coprocessor which is available only for companies that have MFI certification. But it seems that some Chinese companies have a workaround. 

Cheers,

janbrodhaecker, 2020-02-07:
> But it seems that some Chinese companies have a workaround.

If so, this would be amazing! I guess, once the authentication is bypassed, the rest is quite easy to implement (compared to the authentication ;) )

fboulegue, 2020-02-22:
damn nice project if we can support you somehow let me know!

lodi12, 2020-03-23:
> Hi, Unfortunately we are still in early stage. The most important and interesting part is to establish BLE connection between iPhone and Raspberry which is iAP2 authenticated. Normally you need [...]

As far as I know, now Apple MFI uses software authentication but I cannot find any documentations online. To obtain the documentations for CarPlay you must enroll the MFi Program (https://developer.apple.com/programs/mfi/).

janbrodhaecker, 2020-03-23:
There is no proper source on the fact, that Apple uses software authentication from now on. The only thing I could find out is, that HomeKit devices do not need an hardware authentication any longer. For all other devices, there is still the hardware authentication (afaik).

Hakim-Bou, 2020-03-23:
I think we should reverse the iOS frameworks source code of the mfi protocol (using ida/hopper/radare2).
You can find the binaries here (of the xcode simulator):
```
/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/System/Library/PrivateFrameworks/
/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/System/Library/Frameworks/
```

Imo the best solution is to make an iOS tweak (require device jailbroken) to bypass the MFi authentication, but that's difficult too. I already lost so much time trying to do this

AngrySKL, 2020-04-09:
Hi! First i wanna say this is a great project, it give me the ability to build my own wireless carplay. But to be honest, i'm still a little not sure about what should I do. So the basic idea is transfer carplay data through wifi but do wifi connection through bluetooth right? So if I have a rip3 chip with wifi and bluetooth module and configure wifi on AP mode -> connect my iphone to the bluetooth of the rip3 chip and then this project will take care of all the rest things automaticlly?

savent404, 2020-06-21:
If u guys wanna do this on RPi3, crash the MFi first.

WifiHero91, 2020-07-23:
Check this guys for MFI hack.: [Link to PPT http://2013.hackitoergosum.org](http://2013.hackitoergosum.org/presentations/Day3-04.Hacking%20apple%20accessories%20to%20pown%20iDevices%20%E2%80%93%20Wake%20up%20Neo!%20Your%20phone%20got%20pwnd%20!%20by%20Mathieu%20GoToHack%20RENARD.pdf)

sa7mon, 2020-11-08:
Hi all - just checking in. Has anyone got this working with any Raspberry Pi models? I see there are binaries in the repo. If they do work can someone post some general instructions on using them? Thanks!

maaiika, 2020-11-16:
hey，u can buy a carplay dongle 。thats much easy https://github.com/electric-monk/pycarplay

TimorYang, 2021-01-12:
> Check this guys for MFI hack.: [Link to PPT http://2013.hackitoergosum.org](http://2013.hackitoergosum.org/presentations/Day3-04.Hacking%20apple%20accessories%20to%20pown%20iDevices%20%E2%80%93%20Wake%20up%20Neo!%20Your%20phone%20got%20pwnd%20!%20by%20Mathieu%20GoToHack%20RENARD.pdf)
> 
> Other things.: If you use BLE no MFI needed.

According to the document, you need to connect via Bluetooth, and then transmit the WI-FI license certificate through IAP2 protocol to connect to Wi-Fi.
SO, We can’t get around IPA2.

rgoggy, 2021-02-16:
Any update here?
I would like to have wireless car play, but the dongles always produce an own WiFi. So I can not use the wifi of the car. I would like to have the rpi as Converter that uses the same wifi

janbrodhaecker, 2021-02-16:
there is currently no official way to authenticate the raspberry against the iPhone ... so IAP2 is not possible.

cuongnv, 2021-03-18:
Anything update on this project?

CineXMike, 2021-08-14:
> there is currently no official way to authenticate the raspberry against the iPhone ... so IAP2 is not possible.
> 
> still wondering how all the available dongles from china can handle this ...

Something they probably are never gonna share. I wish the good old days where things would be open sourced and could be build on top off.

hbouhadji, 2021-11-02:
[image]
I just got one from eBay, but it's even smaller than I expected... I don't think I can do any soldering

Luccifer, 2021-11-09:
https://github.com/ludwig-v/wireless-carplay-dongle-reverse-engineering

mcr-ksh, 2022-02-14:
> MFI CPIC
What are keywords to look for on ali? (just MFi and CPIC is way too generic)
Anyone found some I2C ready boards to attach to the RPi?

hbouhadji, 2022-07-30:
[image]

I got it working by making an iOS tweak (jailbroken device required) to bypass the mfi authentication and tweaking a lot of other stuff on the program. probably can be done with a custom pcb and soldering a mfi chip on it.

mcr-ksh, 2022-07-30:
I can confirm it works with a custom pcb and MFI chip. We got it working that way

ChrisKader, 2022-08-01:
> MFI

Would a custom PCB even be needed? The RPI has GPIO headers that can be used to communicate with it. Further, with the firmware dumps provided by this [repo](https://github.com/ludwig-v/wireless-carplay-dongle-reverse-engineering/) we have what we would need to use a MFI chip with a pi.

ChrisKader, 2022-08-01:
Would it be possible to pass the auth straight to the radio?

hbouhadji, 2022-08-01:
> I can confirm it works with a custom pcb and MFI chip. We got it working that way
> […](#)

nice, but this need some soldering skill which i don't have

ChrisKader, 2022-08-01:
> No idea what you mean with "pass the auth to the radio". You see the post from [qdk0901](https://github.com/qdk0901) above that has a code snippet of the Apple MFi code, which directly references to the i2c device.

So the mfi IC doesn’t care that it’s doing a challenge response for a device it wasn’t programmed for. Makes sense especially since CarPlay works without internet. 

By radio, I mean the head unit. Treat our device as a pass through and act like it’s not even there. If an auth iap2 request comes in, we send it to the head unit and have it reply.

ChrisKader, 2022-08-01:
Since the head unit is CarPlay ready, it has the IC and such needed to iap2 auth with a phone. When the phone connects to our Pi’s Bluetooth and starts the iap2 procedure, we pass the requests to the CayPlay head unit and in turn pass the replies to the phone.

ChrisKader, 2022-08-01:
[image]

ChrisKader, 2022-08-01:
[image]

ChrisKader, 2022-08-02:
I also like the idea of this: https://github.com/OMGsus/MFiWrapper
Its for controllers but the idea in general is solid.

Also, it looks like we can piggy back off the MFI chip that the cars head unit uses. We can even modify the Capabilities data sent back from the head unit to make it look like it supports bluetooth carplay natively and then use our device as a middle man instead of an accessory.

herenickname, 2022-08-25:
Some Chinese manufacturers of head units have a mfi chip inside and an application for wireless CarPlay.
For example, here is an application from Teyes head unit: https://www.file.io/S4fj/download/E1QzuTsie5NV
Perhaps this will help you understand the issue faster.

Personally, Im interested in getting CarPlay video stream on the screen of my RPi :)
I would like to know what is your progress now? @ChrisKader @qdk0901 @mcr-ksh

Wh1terat, 2023-01-08:
Plenty of MFI ICs available from questionable sources in China.

Paid $245 a few years ago to get 20 boards made up, that included pcb, component sourcing, assembly and shipping.
Not bad given the low quantity.

[image]

liuwentai, 2023-01-08:
尊敬的客户： 您的报装材料已成功发送，我们将尽快审核，请您耐心等待，期间如有任何疑问，请联系您的专属客户经理14795510147。

350d, 2023-03-24:
Any progress here? Minimalistic pcb with MFI chip on looks a good way to start open source wireless dongle project!

davidcorrigan714, 2024-03-18:
@Wh1terat  Got any more info on that board you had made? Trying to find out the pinout and voltages for that chip to make some myself.

Wh1terat, 2024-03-18:
@davidcorrigan714 All available in the leaked IC datasheets. I don't think I've got any local copies as this was a few years ago - but that's where I found all the info I needed back then.

davidcorrigan714, 2024-03-18:
@Wh1terat Thanks! Just found it in the pdfs

Wh1terat, 2024-03-27:
@demianzenkov I never tried to be honest, I was running carplay on a Pi compiled from leaked sources (CarPlay_Communication_Plug-in_R14G17.2)

gsustek, 2024-04-08:
@demianzenkov so, we are helpless without MFI hardware....

demianzenkov, 2024-04-08:
I have one that I can connect to raspberry i2c bus. But this code I guess is for heading side, not the iOS gadget.

demianzenkov, 2024-04-08:
> CarPlay_Communication_Plug-in_R14G17.2

And how it worked for you? You managed to open CarPlay connection with headunit?

demianzenkov, 2024-04-08:
> @demianzenkov,@Wh1terat,    so, we are hopeless without MFI hardware....

I have one that I can connect to raspberry i2c bus. But this code I guess is for headunit side, not the iOS gadget.

Wh1terat, 2024-04-09:
> > CarPlay_Communication_Plug-in_R14G17.2
> 
> And how it worked for you? You managed to open CarPlay connection with headunit?

It was not my goal to emulate a phone, it was to be a carplay receiver (i.e headunit)

djtroubleyin, 2024-12-20:
> 由于主机已准备好支持 CarPlay，因此它具有使用手机进行 iap2 身份验证所需的 IC 等。当手机连接到我们的 Pi 的蓝牙并启动 iap2 程序时，我们会将请求传递给 CayPlay 主机，然后将回复传递给手机。

I found that although you can use a device with an mfi chip to do auth mfi under wifi, if the signature algorithm is inconsistent, it may not be able to bypass auth mfi

signalius, 2026-01-17:
@hbouhadji can you give more details how to configure mdns? I make this repo compilabe on Rpi5 Trixie https://github.com/signalius/WirelessCarPlay but I have problem with configure everything to run the software. I have 3959 connected to Rpi and its working fine ( I tested with wiomoc repo and bluetooth). Any tips?
my discord: @tomsignalius