45clouds/WirelessCarPlay #2: Hi anyone have the build steps? thanks.

liker292, 2020-08-01:
Hi anyone have the build steps? thanks.

Raiden38, 2020-11-05:
Sorry, have to ask the same question!!! I can't figure out how to install your script into my RPI! Is there a guide somewhere or maybe an IMG available? Thanks!

ColdeZhang, 2022-07-25:
I builded it, it comes out with airplayutil and some .so files now I am thinking how to use it...

Here is my build steps (records):

## 编译环境准备

Ubuntu 20.04 ARM64

依赖库：

- libiconv

```shell
wget https://ftp.gnu.org/pub/gnu/libiconv/libiconv-1.17.tar.gz
tar zxvf libiconv-1.17.tar.gz
cd libiconv-1.17
./configure --prefix=/usr/local
make
sudo make install
sudo ln -s /usr/local/lib/libiconv.so /usr/lib/libiconv.so
sudo ln -s /usr/local/lib/libiconv.so.2 /usr/lib/libiconv.so.2
```

- glib-2.0

```
sudo apt-get install libglib2.0-dev
```

- GStreamer

```shell
sudo apt-get install libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools
```

## 编译

### 编译mDNSResponder

执行编译：

```shell
cd mDNSResponder/mDNSPosix
make os=linux
sudo make install
```

### 编译CarPlay

执行编译

```
cd source/PlatformPOSIX
make os=linux
```

BorgoBoy, 2024-08-20:
Hi, have you found a way to make it work?