45clouds/WirelessCarPlay #3: does anyone have full version of carplay.pdf

trywent, 2020-11-18:
need to know carplay data exhange format . wifi Apple Device Information Element (IE) payload

MrPickles2009, 2021-01-19:
is this what you're looking for?
[https://developer.apple.com/carplay/documentation/CarPlay-App-Programming-Guide.pdf](https://developer.apple.com/carplay/documentation/CarPlay-App-Programming-Guide.pdf)

trywent, 2021-01-20:
the link you post may be for app developer. i'm looking for carplay protocol. 
this git repository has provider a protocol file  carpaly.pdf, but is not full version

trywent, 2021-01-20:
> @trywent is this what you're looking for?
> https://developer.apple.com/carplay/documentation/CarPlay-App-Programming-Guide.pdf

the link you post may be for app developer. i'm looking for carplay protocol.
this git repository has provider a protocol file carpaly.pdf, but is not full version

trywent, 2021-01-20:
@MrPickles2009 
the link you post may be for app developer. i'm looking for carplay protocol.
this git repository has provider a protocol file carpaly.pdf, but is not full version

hongfeiyang, 2022-08-22:
> @MrPickles2009
> the link you post may be for app developer. i'm looking for carplay protocol.
> this git repository has provider a protocol file carpaly.pdf, but is not full version

https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

EastArctica, 2023-10-14:
Sorry to bump this, for anyone interested. This is not the same as carplay.pdf, that specification is locked behind mfi so we won't be getting that any time soon 😔. Although the accessory design guidelines are useful as they contain some documentation for things like iap2

liuwentai, 2023-10-14:
尊敬的客户： 您的报装材料已成功发送，我们将尽快审核，请您耐心等待，期间如有任何疑问，请联系您的专属客户经理14795510147。

redssu, 2024-12-10:
@EastArctica Do you know if the PDF that you've linked to is newer than the file in repository? I'm looking for some newer version of this paper, but can't find anything. 

I'm interested in displaying some apps (like maps) on instrument clusters. 

Cheers

EastArctica, 2025-01-01:
@redssu Apologies for the late reply.

To the best of my knowledge it is the same version that is on the github repository. I've looked through CSDN quite a bit looking for any later versions of anything iAP2 or carplay related but wasn't able to find anything at the time.

If you have any luck finding any more information about it please do let me know!