45clouds/WirelessCarPlay #15: Possibly a solution to the mfi problem

jordan679, 2023-09-17:
Have a look at this SDK it might help
https://redpark.com/developer-tools/

liuwentai, 2023-09-17:
尊敬的客户： 您的报装材料已成功发送，我们将尽快审核，请您耐心等待，期间如有任何疑问，请联系您的专属客户经理14795510147。

JJTech0130, 2023-10-25:
> you could also try to decode and use the CarPlay simulator that apple provides and get it running on a rpi because I am guessing there is some way for it to conect to your phone with out any complications but I don’t know how to get it working
I looked into this, but unfortunately this does not work because the new CarPlay Simulator only runs on Macs equipped with a T2 chip, and the simulator uses Apple's BAA authentication based on said chip in place of the MFi coprocessor.

jordan679, 2023-10-26:
ahh ok has anyone looked into the new digital mfi verification I was looking at apple developer on mfi and about 1 year ago apple added digital mfi wich doesn't use any chip