45clouds/WirelessCarPlay #22: Has anyone managed to make this work/build it?

BorgoBoy, 2024-08-20:
I just can't

liuwentai, 2024-08-20:
尊敬的客户： 您的报装材料已成功发送，我们将尽快审核，请您耐心等待，期间如有任何疑问，请联系您的专属客户经理14795510147。

nikit6000, 2024-10-16:
Modify the source code for your purposes and then:
```sh
make -j1 os=linux CC="$(TARGET_CC)" ST="echo" -C "./mDNSResponder/mDNSPosix"
install -D -m 644 "./mDNSResponder/mDNSPosix/build/prod/libdns_sd.so" "$(TARGET_DIR)/usr/lib/libdns_sd.so"
install  -D -m 644 "./mDNSResponder/mDNSShared/dns_sd.h" "$(TARGET_DIR)/usr/include/dns_sd.h"
make -j1 os=linux CC="$(TARGET_CC) -I$(TARGET_DIR)/usr/include -L$(TARGET_DIR)/usr/lib" STRIP="echo" -C "./source/PlatformPOSIX"
```

`$(TARGET_DIR)` - path to your target's sysroot
`$(TARGET_CC)` - path to the gcc for cross compilation