45clouds/WirelessCarPlay #24: Has anyone ever got this working?

AntogamerYT, 2025-01-29:
Hey, I've been thinking about making a wireless dongle for my car (carplay & android auto) with a raspberry pi and i've come across this repo
I understand that there are a lot of technical limitations/issues with a carplay adapter due to the MFi chip being a thing, which, if I've made the right research about it, should be what encrypts the communication between the iphone and the car.

After reading some other issues i realized that these mfi chips are available for really cheap prices on aliexpress, so i was wondering-- would it be possible to buy one of those chips, solder it onto the rpi and actually make the connection/encryption-decryption work? Has anyone ever done this before? Is there a specific type of mfi chip i have to buy or are they all the same? Or am i just dreaming and i can forget about making my own dongle?

liuwentai, 2025-01-29:
尊敬的客户： 您的报装材料已成功发送，我们将尽快审核，请您耐心等待，期间如有任何疑问，请联系您的专属客户经理14795510147。

AntogamerYT, 2025-02-02:
Ok i take back what i said about soldering the chip, apparently it's REALLY small and i never realized it, heres an image from another dongle for reference:

[image]

[image]

(The 5 cents coin is just for reference, it's THAT small)

I could be able to solder this on some kind of breakout board to then interface the chip with the pi, but honestly it's just too complicated for me and i don't think i'll be able to pursue this project, i don't even have an heat gun/the required tools to solder this in the first place (i only have a soldering iron which will 100% NOT work without fucking everything up)

Spadium, 2025-02-08:
If you're worried about desoldering, there are some of them floating around on AliExpress and eBay, just search up "MFI337S3959" but your mileage may vary

Mr-Bossman, 2025-02-17:
@AntogamerYT do you know how to run the program if the IC is added? I can solder one easily and test it. I have been wanting to make the same thing btw.

AntogamerYT, 2025-02-18:
> [@AntogamerYT](https://github.com/AntogamerYT) do you know how to run the program if the IC is added? I can solder one easily and test it. I have been wanting to make the same thing btw.

not this program, it seems to be broken (unless i built it wrong), however i saw [this post](https://wiomoc.de/misc/posts/mfi_iap.html) which seems to go an in-depth reversal/poc to communicate with the chip. I think it would definitely be possible to somehow reimplement the carplay stuff into it

AntogamerYT, 2025-02-18:
> If you're worried about desoldering, there are some of them floating around on AliExpress and eBay, just search up "MFI337S3959" but your mileage may vary

My worry isn't desoldering it, i know about the aliexpress chips, but i'm very worried about soldering one due to their small size (and mainly because i lack the proper soldering skills)

Mr-Bossman, 2025-02-18:
> > [@AntogamerYT](https://github.com/AntogamerYT) do you know how to run the program if the IC is added? I can solder one easily and test it. I have been wanting to make the same thing btw.
> 
> not this program, it seems to be broken (unless i built it wrong), however i saw [this post](https://wiomoc.de/misc/posts/mfi_iap.html) which seems to be an in-depth reversal/poc to communicate with the chip. I think it would definitely be possible to somehow reimplement the carplay stuff into it

Yes, I also saw that looking at the docs, I'm not entirely sure the mfi chip is needed as we can just do a man in the middle type thing. 

If u need one if those chips I can solder one and send it to you.

AntogamerYT, 2025-02-18:
> > > [@AntogamerYT](https://github.com/AntogamerYT) do you know how to run the program if the IC is added? I can solder one easily and test it. I have been wanting to make the same thing btw.
> > 
> > 
> > not this program, it seems to be broken (unless i built it wrong), however i saw [this post](https://wiomoc.de/misc/posts/mfi_iap.html) which seems to be an in-depth reversal/poc to communicate with the chip. I think it would definitely be possible to somehow reimplement the carplay stuff into it
> 
> Yes, I also saw that looking at the docs, I'm not entirely sure the mfi chip is needed as we can just do a man in the middle type thing.
> 
> If you need one if those chips I can solder one and send it to you.
> 
> There is also this repo here [ludwig-v/wireless-carplay-dongle-reverse-engineering#56 (reply in thread)](https://github.com/ludwig-v/wireless-carplay-dongle-reverse-engineering/discussions/56#discussioncomment-12228680) which is reverse engineering a working product

i think mitm would only work if we want to forward a usb connection from iphone -> rpi to the car, since we would theoretically only need to pass the data between the two devices without having to do any of the mfi stuff

however doing a wireless dongle, to my understanding, will need you to create an entirely new ""device"" so that the iphone can pass the data to the dongle wirelessly which will then (the dongle) transfer the data to the car by creating an entirely new connection which will use the usb protocol that the car is expecting

to be honest i don't know if i actually want to continue this, it's definitely possible but in the worse case i would just buy a chip and everything to then not make it work (where are you from? i'm from italy, maybe if i change my mind i'll get back to you)

> [#1 (comment)](https://github.com/45clouds/WirelessCarPlay/issues/1#issuecomment-1200122537)

That looks good, but it won't work for our use case considering that we would still have to handle all the mfi communication between the rpi and the car

Spadium, 2025-03-27:
You could technically have the car/headunit handle the authentication if it supports wired CarPlay, you'd have to accept the radio's authentication and pass over the key that the radio sends from the Pi/other SBC to the wireless carplay connection, but i'm not really sure if its feasible, or even possible. This is just speculation based on the Accessory Authentication document

Mr-Bossman, 2025-03-28:
> You could technically have the car/headunit handle the authentication if it supports wired CarPlay, you'd have to accept the radio's authentication and pass over the key that the radio sends from [...]

I was thinking that too but as far as I can tell it's not possible by looking at the docs. If it was there wouldn't be tons of dongles that have a MFI chip. Also if I recall correctly the repo I linked earlier has talks about the same thing.

Also why is there like an order receipt or something in this thread LOL.

signalius, 2026-01-17:
@AntogamerYT @Mr-Bossman any success with this repo? I have some progress with Rpi 5 Trixie and chinese MFI CP2.0 chip is working. 
Here is my results: https://github.com/signalius/WirelessCarPlay
I also make some code by hand and I can connect to Car HeadUnit by USB and establish cdc_ncm (IP over USB) connection but I cannot stream anything to the HeadUnit.